G6EJD/ESP32-e-Paper-Weather-Display #255: Error in common.h ?

schroepfer, 2024-04-25:
Hello,
i'm using your code since many years and i wanted to update it for the new OWM API 3.
I use Arduino 2.3.2 and installed the source-zip in the library.

With your example Waveshare_1_54.ino i get the following errors:

`In file included from C:\Users\Jürgen\AppData\Local\Temp\.arduinoIDE-unsaved2024325-24144-1brrv0c.g6s1\Waveshare_1_54\Waveshare_1_54.ino:96:
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h: In function 'bool obtain_wx_data(WiFiClient&, const String&)':
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:3: error: expected primary-expression before '?' token
   ?lat={lat}&lon={lon}
   ^
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:4: error: 'lat' was not declared in this scope
   ?lat={lat}&lon={lon}
    ^~~
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:4: note: suggested alternative: 'stat'
   ?lat={lat}&lon={lon}
    ^~~
    stat
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:13: error: expected ':' before '&' token
   ?lat={lat}&lon={lon}
             ^
             :
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:14: error: 'lon' was not declared in this scope
   ?lat={lat}&lon={lon}
              ^~~
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:118:14: note: suggested alternative: 'log'
   ?lat={lat}&lon={lon}
              ^~~
              log
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:123:5: error: 'uri' was not declared in this scope
     uri += "&cnt=" + String(max_readings);
     ^~~
c:\Users\J�rgen\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:126:34: error: 'uri' was not declared in this scope
   http.begin(client, server, 80, uri);
                                  ^~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\libraries\WiFi
  Not used: C:\Users\Jürgen\Documents\Arduino\libraries\WiFiNINA
exit status 1

Compilation error: exit status 1`

Do you have any idea?

Regards
Juergen

G6EJD, 2024-04-25:
I have corrected the issue and also modified the code to match the Arduino JSON v7 requirements
I suggest you go to the repository and copy common.h in the src folder and past that into your version/copy in the libraries folder , or just download this repository again.
I'm not sure if API 3.0 is ready yet.

closed 2024-04-25 by G6EJD

schroepfer, 2024-04-25:
Thank you!
I will try IT.
API 3.0 worked today.

Regards
Juergen

G6EJD, 2024-04-25:
Yes, I've just tried API 3.0, they currently support the cnt= option, but I doubt they will after June'24, that will need some code revisions.

schroepfer, 2024-04-26:
Thank you!
With your new common.h it compiles and worked on my 1.54 ePaper.

PS: german umlauts are not displayed.

Regards
Juergen (in Germany)

G6EJD, 2024-04-26:
My mistake, that’s because the 1.54 variant does t use the u8g library which does support accents.