G6EJD/ESP32-e-Paper-Weather-Display #256: OneCall updates

mrmike8887, 2024-05-19:
I did try to update to the new onecall API, but received all 0's for a reply.  I did decipher some of the data coming back, and making changed to common.h for this, but I am not proficient enough to finish.  here is what I have done so far:
under  
" if (Type == "weather") { "
the roots had to change, some from main to current, or some root had to be added
under the forecast, change root["list"] to root["hourly"]
I am having issues how to deal with forecasts with the new api

[common.txt](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/15370720/common.txt)

mrmike8887, 2024-05-19:
I now get most responses on the serial monitor with the exception of rain.  Cleaned up and deleted my api key
[common.txt](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/15371007/common.txt)

G6EJD, 2024-05-20:
Well done, all that looks OK.
Rain is in the API, but if there's none it's omitted, is that what your seeing.
The differences aren't that great.

mrmike8887, 2024-05-20:
It was raining most of the day yesterday.  I wasn't getting any reading on the serial monitor.
Attached is a text file of the different responses.  It's weird
[debug.txt](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/15380643/debug.txt)

G6EJD, 2024-05-20:
For some reason the json library decoder isn’t always converting some variables to type float, so this accounts for the 0 values, it’s almost certainly choosing type integers instead of floats. I’ll look at the json source code to see what’s happening.

G6EJD, 2024-05-22:
You’ll need to use the directives .as<float>(); to convert some of the json data to the correct variable type. So for the two errant rain and s is types use .as<float>(); or .as<int>();

I see the OneCall does not provide hourly high/low temperatures and only 48-hours of the rest, plus daily values are limited to 7-days, so actually quite a lot of change and a lot has to change. I’m still thinking through how to make the transition as easy as possible. 
also, it’s an https / secure client call, so not as straightforward anymore. Plus I’m seeing memory is very short now because the json response is huge to get the same data, not that smart a change really, I think they were too obsessed with mirroring the DarkSky API, then DarkSky was purchased by Apple snd access was removed from their API, so it turned out to be a somewhat futile change with no apparent benefit!

closed 2024-10-11 by G6EJD