G6EJD/ESP32-e-Paper-Weather-Display #257: Waveshare REV2.3 driver board won't refresh

Daniel-Busselot, 2024-05-25:
I can't get my display to update with the new 2.3 driver board, I have PWR connected to 3.3V 
I'm using an ESP32 DEV board 
here's my pin out for the display

BUSY = 25;
CS   = 15;
RST  = 26; 
DC   = 27; 
SCK  = 13;
MISO = 12; 
MOSI = 14;

G6EJD, 2024-05-28:
So you defined the pins like this:
static const uint8_t EPD_BUSY = 25;  // to EPD BUSY
static const uint8_t EPD_CS   = 15;  // to EPD CS
static const uint8_t EPD_RST  = 26; // to EPD RST
static const uint8_t EPD_DC   = 27; // to EPD DC
static const uint8_t EPD_SCK  = 13; // to EPD CLK
static const uint8_t EPD_MISO = 12; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14; // to EPD DIN

Daniel-Busselot, 2024-05-29:
Correct, If i load an example code for the waver share display it works fine. Edit the pins to what the wave share code is using and the display doesn't update  
I've got an ESP-Wroom-32

Does the code need to  be updated for the newer display driver board?

G6EJD, 2024-05-29:
Which driver is the Waveshare code example using?
This code uses the GxEPD2 driver and what size screen are you using? Your s teen type has to be supported by the EPD2 library

Daniel-Busselot, 2024-05-30:
Here's the display i have 
https://www.aliexpress.us/item/3256805935116162.html?spm=a2g0o.productlist.main.1.37c43432oQgkch&algo_pvid=51f4c525-c081-4268-9407-f2af2fb10b36&algo_exp_id=51f4c525-c081-4268-9407-f2af2fb10b36-0&pdp_npi=4%40dis%21USD%21183.99%21165.59%21%21%21183.99%21165.59%21%402101e5f117170365944496211e3c9f%2112000035848541796%21sea%21US%212343225675%21&curPageLogUid=C9HM2ddIGWqI&utparam-url=scene%3Asearch%7Cquery_from%3A

my Example code is the GxEPD2 library
Using display driver. GxEPD2_DRIVER_CLASS GxEPD2_1160_T91
960x680

G6EJD, 2024-05-30:
The reason that display won’t work is the huge amount of RAM required to build the image and then to transfer it. 
it needs special techniques to get it displaying and the sheer size may preclude it working all-together.
I’ll have a look at it today to see if there might be a solution.

G6EJD, 2024-05-30:
Which example code are you using and have you enabled the half-buffer option in the GxEPD2 driver?

Daniel-Busselot, 2024-05-31:
Here's the code i used 

[GxEPD2_WS_ESP32_Driver.zip](https://github.com/user-attachments/files/15507950/GxEPD2_WS_ESP32_Driver.zip)

Daniel-Busselot, 2024-06-12:
Still can't get the display to update, I'm using Waveshare 9.7
Here's the code
[Waveshare_9_7.zip](https://github.com/user-attachments/files/15799396/Waveshare_9_7.zip)

G6EJD, 2024-06-19:
I've now how chance to look through your code and cannot see any issues with it at all.

So are you saying it displays correctly with the GxEPD2 example for the 9.7" display?

If so, then the issue (assuming the pin connections are identical) must be in the screen initialisation, and so you should inspect the difference in the initialisation lines of my code verses those of GxEPD2.

closed 2024-10-11 by G6EJD