G6EJD/ESP32-e-Paper-Weather-Display #261: Attempting with XAOI Seed ESP32C6

technash85, 2024-08-18:
Hello,

Thanks for the very attractive project and for keeping up with the API changes. As title states, I am attempting to drive Waveshare 400x300 4.2i e-ink display V2 via SEED ESP32C6. I thought this would be the correct wiring (can see output in Arduino IDE where it is communicating with API and getting replies but no screen output):

BUSY = 3
CS = 0
RST= 2
DC=1
SCK = 8
MISO = 9 but not connected 
MOSI = 10

ANY help -- or pointing in the right direction even -- is much appreciated.

G6EJD, 2024-08-18:
Using Dn GPIO pin numbering is only going to work if you have selected the correct board type in the Arduino IDE, which for your board must be the XIAO ESP32C6, then:
BUSY = D3 or GPIO-21
CS = D0 or GPIO-0
RST= D2 or GPIO-02
DC = D1 or GPIO-01
SCK = D8 or GPIO-19
MISO - not connected
MOSI = D10 or GPIO-18

So you have two approaches for the display object:
// Connections for e.g. XIAO ESP32C6 will only work if you select the board type as XIAO ESP32C6 (maybe)
static const uint8_t EPD_BUSY = D3;  // to EPD BUSY
static const uint8_t EPD_CS   = D0;  // to EPD CS
static const uint8_t EPD_RST  = D2; // to EPD RST
static const uint8_t EPD_DC   = D1; // to EPD DC
static const uint8_t EPD_SCK  = D8; // to EPD CLK
static const uint8_t EPD_MISO = -1; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = D10; // to EPD DIN

Then:

GxEPD2_BW<GxEPD2_420, GxEPD2_420::HEIGHT> display(GxEPD2_420(/*CS*/ EPD_CS, /*DC*/ EPD_DC, /*RST*/ EPD_RST, /*BUSY*/ EPD_BUSY));

OR:

// Connections for e.g. ESP32C6 using GPIO numbers
static const uint8_t EPD_BUSY = 21;  // to EPD BUSY
static const uint8_t EPD_CS   = 0;  // to EPD CS
static const uint8_t EPD_RST  = 2; // to EPD RST
static const uint8_t EPD_DC   = 1; // to EPD DC
static const uint8_t EPD_SCK  = 19; // to EPD CLK
static const uint8_t EPD_MISO = -1; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 18; // to EPD DIN

Then:

GxEPD2_BW<GxEPD2_420, GxEPD2_420::HEIGHT> display(GxEPD2_420(/*CS*/ EPD_CS, /*DC*/ EPD_DC, /*RST*/ EPD_RST, /*BUSY*/ EPD_BUSY));

You may also need to try a different display object for your screen type, like this:

technash85, 2024-08-18:
Thanks so much! Will give it a whirl this evening so time and report back. What you've written makes perfect sense! I'm still pretty new to the SEED versions of ESP32 and this cleared quite a few things up for me about pin numbering in general on them.

technash85, 2024-08-19:
K. Gave the first part of it a try. I'm getting this error when trying to compile now (sorry if it is something obvious and I really appreciate you writing back to me!): 

`C:\tools\Arduino\eWeather\Waveshare_4_2\Waveshare_4_2.ino:73:62: error: expected primary-expression before '/' token
   73 | GxEPD2_BW<GxEPD2_420, GxEPD2_420::HEIGHT> display(GxEPD2_420(/CS/ EPD_CS, /DC/ EPD_DC, /RST/ EPD_RST, /BUSY/ EPD_BUSY));`

Am I supposed to plug in the values or write it just like you've written it in your reply?

G6EJD, 2024-08-19:
For some reason the comment asterisks have been striped out so when you see /RST/ change to /* RST */ or just remove leaving the PIN numbers or names 
Then /*CS*/ EPD_CS, becomes   EPD_CS, EPD_DC, and so-on

technash85, 2024-08-24:
I need to do some more reading on the display library to wrap my head around it some more. I really thank you for all your help! I'm having no luck getting the display to work. I have the Waveshare Rev. 2.2. I do see the code runs, but I think I'm seeing an error come across the serial monitor just before it enters deep sleep. 

`21:07:37.781 -> E (3795) adc_common: adc_io_to_channel(28): invalid gpio number`

G6EJD, 2024-08-24:
ADC channels can only be on GPIO 32 or above

its most likely your pin definitions for the display connection that are wrong.

technash85, 2024-08-24:
Would they be SS, 1, 2 and 21? Maybe I'm not understanding something. haha. I apologize.

G6EJD, 2024-08-24:
The pins you were originally quoting I can now see are wrong for the display connections, the XAIO pins need to be redefined. Search for the C6 variant and make sure you use the MOSI MISO and CLK as a minimum so that most of the SPI bus is correct

G6EJD, 2024-08-24:
For the ADC you can use GPIO-1, 2, 3, 4, 5, 6

technash85, 2024-08-24:
Thanks for your help! Got it working now!

G6EJD, 2024-08-25:
Good bews

closed 2024-08-25 by G6EJD