G6EJD/ESP32-e-Paper-Weather-Display #262: Need an independent API-Call Function

Zurrmaxe, 2024-08-20:
Hello Friends
I also use the display for my solar display. 
To do this, WiFi and the ESP32 must not be switched off and word 24 Hours. 
To do this, I switched off all the SLEEP functions in this ePaper-code.

But now, the weather update doesn't work anymore.

Now i want to start the weather API manually every 30 minutes with my own code
Therefore i need  the "codeword" or StringText or first step  to start the weatherupdate once.
I can handle the rest (Timer , if then else ...pp) of the code.
Alfred

G6EJD, 2024-08-24:
You can’t do that, just make the ESP32 go to Deep Sleep then set to same 30-mins later or over night for the required number of hours, so definite two timer values  30-mins snd 8-hours or whatever you need.

Zurrmaxe, 2024-08-25:
I have an 7inch Display and want to show the weatherstation, the power of my solarpanels and the power consumtion of my house.
So the esp32 and the WiFi.init are always "on" to get the JSON-Files from the solar-stations and the shelly EM every 2 to 4 seconds.  In this case, the weatherstation get no OWM updates. I must resart the OMW manualy like this:

void timestart() {
    //ESP Restart every 01 and 31 Minutes of an houre
    if(timeClient.getMinutes() == OnMin){
     // DisplayWeather();
      ESP.restart();
    }
      else if (timeClient.getMinutes() == OnMinA){
         //DisplayWeather();
         ESP.restart();
    }

}

But then, the Screen always flickers when the ESP32 wake up or restart .
So im looking for a code to start only the OWM update without starting the ESP.

G6EJD, 2024-08-25:
Just make sure the WiFi is connected, then:
WiFiClient client;   // wifi client object
      while ((RxWeather == false || RxForecast == false) && Attempts <= 2) { // Try up-to 2 time for Weather and Forecast data
        if (RxWeather  == false) RxWeather  = obtain_wx_data(client, "weather");
        if (RxForecast == false) RxForecast = obtain_wx_data(client, "forecast");
        Attempts++;
      }
      if (RxWeather && RxForecast) { // Only if received both Weather or Forecast proceed
        StopWiFi(); // Reduces power consumption
        DisplayWeather();
        display.display(false); // Full screen update mode
      }

That will get new data and display it. BUT be warned, OWM has introduced since yesterday strict controls on API access limited to 1000/day after which you have to pay for access, so make a request more than every 2-mins and they will block your account or charge you. You need to sign up for a new API key too.

Zurrmaxe, 2024-09-22:
[image]

Good evening
Im working on a projekt, to convert your weatherstaion to a colorfull 7-Inch-TFT-Dissplay. 
The display will later have various functions. So far I have integrated my two solar displays (2x Hoymiles) and the power consumption (Shelly 3EM). Also an interior temperature display.
It is not finished yet and is still under construction.

73 Alfred DK1AW

G6EJD, 2024-09-22:
That’s a really good idea, something I have never considered with a colour TFT. Thanks for the idea, as I can add my solar output to the display together with battery storage level too. 73 Dave G6EJDill close the issue but we can continue chatting on this thread

closed 2024-09-22 by G6EJD

Zurrmaxe, 2024-09-23:
Hi Dave
I wann to have only one display in my kitchen. Thats my idea.
In  a week, i will send you my code.
73 Alfred

Zurrmaxe, 2024-10-03:
Hi Dave
You can look on my repository. There are my first versions of the Color-Display with the solar-controll.
I created a wlan sensor with a ESP32-Wroom to connect lots of sensors to the display. 
DHT11 and BME280 are now working.
I did not use ESP_NOW, because it is not so good as WLAN (for longer distance).
Alfred

G6EJD, 2024-10-03:
Thanks Alfred, I shall try that later after work. I have the Waveshare 7” version so I will need to change some of the screen objects, you’ve done a good job.  David

Zurrmaxe, 2024-10-03:
The display is of course constantly online because of the queries.
However, the weather data and weather displays are only updated when the program is started !
I have therefore modified the program so that it always performs a reset (ESP.deepSleep(1e6)) every 30 minutes (59.1 and 30.1).
Normally around 48 inquiries are made per day.
You could also set it to every 15 minutes.
Depending on your wallet. 
Since the sensor queries go via WLAN and http.get, other local sensors can also be integrated. I'm thinking about wind speed, rain and sun. All you have to do is expand the code.
It would also be possible to send an http.post signal to the sensor to trigger switching signals.
So when it rains, the sun blind is retracted.
I have a few ideas but little time at the moment. Maybe other users can continue with this. 
The whole thing was just supposed to be a test of the 7-inch display and not a finished product.
It still has small defects that will be corrected in the near future.

73 Alfred