G6EJD/ESP32-e-Paper-Weather-Display #263: Compilation Error

rg373, 2024-09-10:
Hello,
during Compilation i get following error:

c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `mm_to_inches(float)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:3: multiple definition of `mm_to_inches(float)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:3: first defined here
c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `hPa_to_inHg(float)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:8: multiple definition of `hPa_to_inHg(float)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:8: first defined here
c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `JulianDate(int, int, int)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:12: multiple definition of `JulianDate(int, int, int)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:12: first defined here
c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `SumOfPrecip(float*, int)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:26: multiple definition of `SumOfPrecip(float*, int)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:26: first defined here
c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `TitleCase(String)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:34: multiple definition of `TitleCase(String)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:34: first defined here
c:/users/aa/appdata/local/arduino15/packages/esp32/tools/esp-xs2/2302/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\src\common_functions.cpp.o: in function `NormalizedMoonPhase(int, int, int)':
C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7\src/common_functions.cpp:43: multiple definition of `NormalizedMoonPhase(int, int, int)'; C:\Users\aa\AppData\Local\Temp\arduino\sketches\B0894E96F06A947D53593A74C88A286F\sketch\common_functions.cpp.o:C:\Users\aa\Downloads\ESP32-e-Paper-Weather-Display-master\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5_T7/common_functions.cpp:43: first defined here
collect2.exe: error: ld returned 1 exit status

in the Code ( Waveshare 7_5_T7) i use:

int JulianDate(int d, int m, int y) {
  int mm, yy, k1, k2, k3, j;
  yy = y - (int)((12 - m) / 10);
  mm = m + 9;
  if (mm >= 12) mm = mm - 12;
  k1 = (int)(365.25 * (yy + 4712));
  k2 = (int)(30.6001 * mm + 0.5);
  k3 = (int)((int)((yy / 100) + 49) * 0.75) - 38;
  // 'j' for dates in Julian calendar:
  j = k1 + k2 + d + 59 + 1;
  if (j > 2299160) j = j - k3; // 'j' is the Julian date at 12h UT (Universal Time) For Gregorian calendar:
  return j;
}

thanks for your help

G6EJD, 2024-09-10:
You need to remove all the functions in your Sketch folder leaving just the .ino credentials etc. All the functions come from the library.

G6EJD, 2024-09-11:
You need to set the display object line to match your display size/type, which is?

rg373, 2024-09-12:
Everything fine now. Thank you

closed 2024-10-11 by G6EJD