G6EJD/ESP32-e-Paper-Weather-Display #265: No 3 hrs forecast

austinroolon, 2024-10-05:
Display:
2.9 WeActStudio Epapaer
I have only daily forecast (today+2 days).
I prefer 3 hrs forecast, but I can`t find how to do it (like on pictures). Something change in sketch? 

BTW it is important to change ePaper display definition
// 2.9'' EPD Module
//GxEPD2_BW<GxEPD2_290_BS
//GxEPD2_3C<GxEPD2_290_C90c

G6EJD, 2024-10-05:
There are many 2.9” displays on the market, so the correct type needs to be selected from the GxEzPD2 library.

OWM only publish 3-hour forecast data , look at 
Void DisplayWeather()
In that function is a count to receive the forecasts 0, 1 and 2 so you could increase the 2 to 3

austinroolon, 2024-10-05:
In my shetch in Void DisplayWeather() section I have:
int Forecast = 2, Dposition = 0;  (no changes)
any on my display I have DAILY forecast only. At bottom line there are 4 fields:
SUN MON TUE and the big one with Moon and sunrise/sunset time.
but 
I have no idea how change into:
NOW/ NOW+3hrs / NOW+6hrs

austinroolon, 2024-10-07:
I used Waveshare_2_9_T5 instead of before used Waveshare_2_9_v4 and it is ok
Waveshare_2_9_v4 is a daily update sketch
Waveshare_2_9_T5 is 3-hrs update sketch.
Solved

closed 2024-10-07 by austinroolon