G6EJD/ESP32-e-Paper-Weather-Display #266: ESP32-Wroom-32 - ePaper 2.9

AntoszHUN, 2024-10-11:
Hello! We managed to build the weather station. Very cool! But I have 2 problems. One is that I can't see the Wi-Fi signal. I think it would fit in the upper part of the display. My other problem is that I can't see the battery charge. How can I call this? and the ADC battery monitor doesn't want to work either. If I plug in the voltage divider and the ADC gets 3.3V, the ESP32 doesn't work. Can I ask for help?

AntoszHUN

Thank you in advance!

[image]

G6EJD, 2024-10-11:
It won’t draw WiFi signal strength if that’s what you mean as there is no function in the code to do that, to do so you’d need to include a copy of the function from another code variant to display it somewhere on the top-right of the screen. 

The battery monitor won’t work unless you wire a voltage divider (assuming one is not fitted to your ESP32 board ) to pin 35 or you change the 35 pin number in the code in the battery reading function to the pin your using.

AntoszHUN, 2024-10-12:
> It won’t draw WiFi signal strength if that’s what you mean as there is no function in the code to do that, to do so you’d need to include a copy of the function from another code variant to display [...]

I have gone through all these steps. The ESP32-Wroom-32 may have one of the pins already installed with a voltage divider? How do I know this?

AntoszHUN, 2024-10-12:
Which lines should be copied and from where? I want to see the wi-fi signal strength.

AntoszHUN, 2024-10-13:
How can I set the first day of the week to Monday?

G6EJD, 2024-10-13:
For battery measurement you’ll need to check your board suppliers details it may not have a voltage divider

G6EJD, 2024-10-13:
void DrawRSSI(int x, int y, int rssi) {
  int WIFIsignal = 0;
  int xpos = 1;
  for (int _rssi = -100; _rssi <= rssi; _rssi = _rssi + 20) {
    if (_rssi <= -20)  WIFIsignal = 20; //            <-20dbm displays 5-bars
    if (_rssi <= -40)  WIFIsignal = 16; //  -40dbm to  -21dbm displays 4-bars
    if (_rssi <= -60)  WIFIsignal = 12; //  -60dbm to  -41dbm displays 3-bars
    if (_rssi <= -80)  WIFIsignal = 8;  //  -80dbm to  -61dbm displays 2-bars
    if (_rssi <= -100) WIFIsignal = 4;  // -100dbm to  -81dbm displays 1-bar
    display.fillRect(x + xpos * 5, y - WIFIsignal, 4, WIFIsignal, GxEPD_BLACK);
    xpos++;
  }
  display.fillRect(x, y - 1, 4, 1, GxEPD_BLACK);
  drawString(x + 5,  y + 5, String(rssi) + "dBm", CENTER);
}

Then in the code:
DrawRSSI(x, y, RSSI); 

check the 7.5” code for how to get WiFi RSSI value

closed 2024-10-13 by G6EJD

AntoszHUN, 2024-10-13:
I've come this far. Thanks for your help. But there is no reason to close this topic. None of my problems have been solved. I figured if I asked the developer, I could get help. But I see I can't really count on that here. Thank you.

AntoszHUN, 2024-10-13:
> For battery measurement you’ll need to check your board suppliers details it may not have a voltage divider

No. There may be. I did it. But it still doesn't work.

AntoszHUN, 2024-10-13:
I copied the commands. But the placement is a bit opaque. I don't understand all the interesting notations x +5 and y - etc... I don't understand where they are placed on the display.

[image]

G6EJD, 2024-10-13:
X and y are the screen locations where you wan filter display to be. In the function everything that is drawn is relative to the x and y location. Try it at the middle the screen first such as 150, 100 then adjust X and y to locate it somewhere suitable,

G6EJD, 2024-10-13:
I'm still answering your issues!

AntoszHUN, 2024-10-14:
> I'm still answering your issues!

Okay. Thank you!

AntoszHUN, 2024-10-14:
The display has a resolution of 296x128. The middle of it is 148x64. But the signal strength is not displayed.

[image]

AntoszHUN, 2024-10-14:
I appreciate that you have prepared this perfect weather programme. I really like it. I have slightly modified the layout of the 2.9" display. However, I use this ESP32-Wroom-32 development board. The battery ADC does not want to work. If I apply 3.3V from the voltage divider to GPIO35, the ESP32 will not start. Only the red LED lights up and does nothing. It's like the ESP32 freezes. 

[image]
[image]

G6EJD, 2024-10-14:
WiFi - you have to call the function for it to be displayed!
 DisplayWeather();
 DrawRSSI(120,100, WiFi.RSSI());
display.display(false); // Full screen update mode

For the battery voltage display you need to use not D35 on your board but VP (which is the ESP32 GPIO36) so you need to change the ADC reading line to 36 in the display battery function.

AntoszHUN, 2024-10-14:
> WiFi - you have to call the function for it to be displayed! DisplayWeather(); DrawRSSI(120,100, WiFi.RSSI()); display.display(false); // Full screen update mode
>
> For the battery voltage display you [...]

Super. The test is good. But why no reading? Is there somewhere to change the size? 

[image]

G6EJD, 2024-10-14:
You may need to get the RSSI value first before the WiFi is turned off to save power.
At the top of the programme create a variable called WiFi_RSSI, then after the WiFi has started add
WiFi_RSSI = WiFi.RSSI();

Then call DrawRSSI(x, y, WiFi_RSSI);

AntoszHUN, 2024-10-14:
> You may need to get the RSSI value first before the WiFi is turned off to save power. At the top of the programme create a variable called WiFi_RSSI, then after the WiFi has started add WiFi_RSSI = WiFi.RSSI();
> 
> Then call DrawRSSI(x, y, WiFi_RSSI);

And where do I enter this? Which line?

G6EJD, 2024-10-14:
After line 119

WiFi_RSSI = WiFi.RSSI();

AntoszHUN, 2024-10-15:
> After line 119
> 
> WiFi_RSSI = WiFi.RSSI();

It says it is not declared.
[image]

G6EJD, 2024-10-15:
As I said earlier at the top of the programme create a variable called WiFi_RSSI, with:

int WiFi_RSSI;

then after the WiFi has started add WiFi_RSSI = WiFi.RSSI();

These are basic C requirements

AntoszHUN, 2024-10-15:
> As I said earlier at the top of the programme create a variable called WiFi_RSSI, with:
> 
> int WiFi_RSSI;
> 
> then after the WiFi has started add WiFi_RSSI = WiFi.RSSI();
> 
> These are basic C requirements

Ohhh sorry. I wasn't paying attention. I forgot that. 🤦🏼

AntoszHUN, 2024-10-15:
> As I said earlier at the top of the programme create a variable called WiFi_RSSI, with:
> 
> int WiFi_RSSI;
> 
> then after the WiFi has started add WiFi_RSSI = WiFi.RSSI();
> 
> These are basic C requirements

This does not want to work...
[image]
[image]

G6EJD, 2024-10-15:
Nice line 119 to 121 so that it's after WiFi start

Replace Draw_RSSI WiFi.RSSI() with the variable WiFi_RSSI which got a value on line 121

AntoszHUN, 2024-10-15:
> Nice line 119 to 121 so that it's after WiFi start
> 
> Replace Draw_RSSI WiFi.RSSI() with the variable WiFi_RSSI which got a value on line 121

I do not understand..

G6EJD, 2024-10-15:
Move your line 119 to be 121 and delete WiFi.RSSI() with WiFi_RSSI

AntoszHUN, 2024-10-15:
> Move your line 119 to be 121 and delete WiFi.RSSI() with WiFi_RSSI

The display still shows 0 dBm.
[image]

G6EJD, 2024-10-15:
You still haven't assigned a value to WiFi_RSSI like this:
WiFi_RSSI = WiFi.RSSI();
And move the line 122 so that it is after the command start_wifi on line 123 in your code version.

Next you haven't replaced WiFi.RSSI() in the call to DrawRSSI on line 142 so that it's WiFi_RSSI

AntoszHUN, 2024-10-16:
> You still haven't assigned a value to WiFi_RSSI like this: WiFi_RSSI = WiFi.RSSI(); And move the line 122 so that it is after the command start_wifi on line 123 in your code version.
> 
> Next you haven't replaced WiFi.RSSI() in the call to DrawRSSI on line 142 so that it's WiFi_RSSI

Veri nice! Ohh great. It works now. Thank you! Now I just need to check the battery status, but there are several problems.

AntoszHUN, 2024-10-22:
Hi!
I have purchased NodeMCU ESP-32s modules. I loaded the weather station as needed. ADC input GPIO36 which is the SenseVP. However, if I connect the ADC input to the voltage divider and connect the battery 5V and GND to the input, the ESP32 does nothing. What can I do? I have tried 3 ESP32's. The phenomenon is the same for all of them.

G6EJD, 2024-10-22:
So you should be connecting the voltage divider and ADC like this:

5v
|
R1
+ ESP32 GPIO36
R2
|
Gnd

Where R1 is typically 100K and R2 100K

And you need to change the PIN number used for analogread in battery measurement function

AntoszHUN, 2024-10-23:
> So you should be connecting the voltage divider and ADC like this:
> 
> 5v | R1
> 
> * ESP32 GPIO36
>   R2
>   |
>   Gnd
> 
> Where R1 is typically 100K and R2 100K
> 
> And you need to change the PIN number used for analogread in battery measurement function

That's exactly what I did. But ESP32 won't wake up. The red LED is on. But the ESP32 does nothing. It does not wake up and it does not update the display.

AntoszHUN, 2024-10-23:
Are you sure all this is enough to write out the battery data? Because other displays have different battery settings in the INO file. E.g. for the 4.2 display..

//#########################################################################################
void DrawBattery(int x, int y) {
  uint8_t percentage = 100;
  float voltage = analogRead(36) / 4095.0 * 7.46;
  if (voltage > 1 ) { // Only display if there is a valid reading
    Serial.println("Voltage = " + String(voltage));
    percentage = 2836.9625 * pow(voltage, 4) - 43987.4889 * pow(voltage, 3) + 255233.8134 * pow(voltage, 2) - 656689.7123 * voltage + 632041.7303;
    if (voltage >= 4.20) percentage = 100;
    if (voltage <= 3.50) percentage = 0;
    display.drawRect(x + 15, y - 12, 19, 10, GxEPD_BLACK);
    display.fillRect(x + 34, y - 10, 2, 5, GxEPD_BLACK);
    display.fillRect(x + 17, y - 10, 15 * percentage / 100.0, 6, GxEPD_BLACK);
    drawString(x + 60, y - 11, String(percentage) + "%", RIGHT);
    //drawString(x + 13, y + 5,  String(voltage, 2) + "v", CENTER);
  }
}
//#########################################################################################

G6EJD, 2024-10-23:
You have added a function call to invoke it? After the line DisplayWeather(); add DrawBattery(x,y); choosing your required x,y values on the screen.
In the serial port you should see it didplay battery voltage from:
Serial.println("Voltage = " + String(voltage));

G6EJD, 2024-10-23:
> That's exactly what I did. But ESP32 won't wake up. The red LED is on. But the ESP32 does nothing. It does not wake up and it does not update the display.

If the ESP32 is during the sleep hours it wont wake up! The hours to wake and sleep are defined in the source code, they can be varied.

AntoszHUN, 2024-10-23:
I try to find a good layout to fit everything. But the display is small. How can the wi-fi icon be smaller?

G6EJD, 2024-10-23:
The WiFi could be just the signal strength in DBM so just comment out the lines that draw the signal icon leaving just the value.

AntoszHUN, 2024-10-23:
> The WiFi could be just the signal strength in DBM so just comment out the lines that draw the signal icon leaving just the value.

It's okay if I can see the field. Just a smaller rubric would be better.

AntoszHUN, 2024-10-23:
Can such a wi-fi icon be created? If so, how?

[image]

G6EJD, 2024-10-23:
It depends on how good your mathematical skills are, you can draw a filled circle at x, y that's easy, but there are no draw commands for an arc, so then you'd need to create an arc command using drawpixel or something like it, draw an arc, then move y up and draw another and so on. Or create an icon and display that.

AntoszHUN, 2024-10-23:
I see. Unfortunately, I'm not as professional as you. Only small programming works for me. I understand what you wrote. I'd love to have such a development.

My goal would be to get it to the top line. Later it will be separated by lines. I would like to place it to the right of the battery. 

[image]

G6EJD, 2024-10-23:
That's why I chose vertical lines drawn with the collect command to get thicker lines, stills gives relative signal strength, you could leave out the line that displays dBm as it doesn't mean much band for the battery just used the filled battery symbol which is represented by % of battery charge remaining.

AntoszHUN, 2024-10-25:
My question is, why is this value in the code?  / 4096.0 * 7.46; The max charge voltage of a battery is 4.20V, is 7.46 not a typo? Just because it doesn't give me the right value. Also, the original values are 0 - 4095 right? Can you help me to understand?

G6EJD, 2024-10-25:
The thing to learn about binary numbering is that 0 is a number so if you count all the numbers between 0-4095 then there are 4096!

Every ESP32 ADC reference is different as they are not what's called laser trimmed in production to equalise them all. Therefore each has to be individually adjusted in the code, the ADC is not a precision type but adequate for 99% of reading accuracy. 

The ADC reading has to be scaled, so the first is to a ratio:
ADC Reading / 4096 or a % it makes no difference.

Next the external voltage divider has to be applied again to scale the result, let's say you use a divide by 2 then the constant would be the ADC maximum scale of 3.3v x 2 or 6.6  but in most cases the voltage divider is set higher to give some spare measurement over 3.3v, say 2.15 then the constant is 7.095

So that's how the calculation is determined.

In reality measure the actual battery voltage, then note the ADC reading and divide the two, then use this result to adjust the constant value say the battery is measured with a DVM at 3.85v and the ADC measures 3.75v then 3.85/3.75=1.0267 now multiple the constant by 1.0267 to increase the reading value to get an exact result. 

The actual calculation of ADC voltage is more more complex because of ADC input currents from the external voltage divider and the internal ADC reference voltage, but for the most part these can be ignored and a single constant gives very accurate results.

AntoszHUN, 2024-10-25:
> The thing to learn about binary numbering is that 0 is a number so if you count all the numbers between 0-4095 then there are 4096!
>
> Every ESP32 ADC reference is different as they are not what's [...]

I have the battery fully charged at 4.176V. This is connected to the ESP32 VIN and GND legs. This is how it works. Also there is a 100-100Kohm voltage divider. But I don't get an accurate result on the epaper display. What do I need to rewrite in the program to get it right? For me, the math was never good.

G6EJD, 2024-10-25:
Yes you need to modify the source code the ADC read line with 7.46 that you mentioned, so change the 7.46 to 
4.17/(your reading) x 7.46

Say your displayed reading is 4.3v then
4.17/4.3x6.46=6.265
Chas get the source code and recompile 

Or, try 7.0 and adjust by trial and error and you may need to adjust a few times.

AntoszHUN, 2024-10-25:
> Yes you need to modify the source code the ADC read line with 7.46 that you mentioned, so change the 7.46 to 4.17/(your reading) x 7.46
>
> Say your displayed reading is 4.3v then 4.17/4.3x6.46=6.265 [...]

Super. Thank you very much. I will try it as soon as I get home after work. Thank you.

AntoszHUN, 2024-10-25:
> Yes you need to modify the source code the ADC read line with 7.46 that you mentioned, so change the 7.46 to 4.17/(your reading) x 7.46
>
> Say your displayed reading is 4.3v then 4.17/4.3x6.46=6.265 [...]

Shouldn't the voltage divider be set so that the 4.2V battery voltage is 3.3V on the GPIO36 pin? Or is the point of the voltage divider to always half the voltage? 4.2V halved to 2.1V?

AntoszHUN, 2024-10-25:
I managed to calculate the exact value. Great. One instrument and the ESP32 shows a good reading. 
[image]
[image]
[image]

G6EJD, 2024-10-25:
Yes you could alter the ratios so that there's 3.3v at the ADC input when the battery is 4.2v, it's a personal choice, but if the battery is ever on charge you might see a voltage above 4.2v, so I always add sone spare measurement capacity. You will find the ADC becomes non-linear above 3.0v so it is best to keep the maximum voltage below 3.0. You can also use a compensatory function called a polynomial that will remove non-linearity in the ADC measurement, but that's getting complex. 

It's all down to a personal preference as to which solution is chosen and used.

AntoszHUN, 2024-10-26:
> Yes you could alter the ratios so that there's 3.3v at the ADC input when the battery is 4.2v, it's a personal choice, but if the battery is ever on charge you might see a voltage above 4.2v, so I [...]

I see. Then it's better as it is. However, there may be a contact fault. Because the settings were good, I charged the battery and the epaper is no longer showing good values. After every update it shows something else. 4.20V later 4.24V then again update and 4.17V but the battery itself is 4.20V voltage I have according to the multimeter.

G6EJD, 2024-10-26:
This is due to what's called quantisation errors produced by the (all) ADC. The reading will tend to  to jitter, yielding +/-1 in 4096 parts, really for the voltage display where is prints out to the screen change ',2' to ',1' so that it displays to one decimal which would be more appropriate for the ADC resolution. Eg 4.17 becomes 4.2

AntoszHUN, 2024-10-26:
> This is due to what's called quantisation errors produced by the (all) ADC. The reading will tend to to jitter, yielding +/-1 in 4096 parts, really for the voltage display where is prints out to the [...]

I see. So this error cannot be prevented. What value of resistor can be used to correct it? I was thinking, based on the 3D printer motherboards, that I would assemble the voltage divider from 2pcs 47Kohm resistors. These have a combined resistance of 100Kohm rounded up. If you take the 5% tolerance for each of the two resistors. I think this is realistic if you take the 5V max input voltage as a basis. This way the battery is not subjected to a high load due to the resistors. Am I right?

G6EJD, 2024-10-26:
You could use 2x470k resistors and get the same result (divide by 2) as the ADC input current is quite low. One easy solution is to add a capacitor across the ADC input and ground as that will remove any input noise. 
An alternative solution is to take say 10 ADC readings in succession and then divide the result by 10 to get an average.
float voltage = 0;
for (int r = 0; r < 10; r++){
  voltage = analogRead(35) / 4096.0 * 7.4;
}

voltage /= 10;

Selecting the ADC pin your using instead of 35 and changing the constant 7.4 too

AntoszHUN, 2024-10-26:
> You could use 2x470k resistors and get the same result (divide by 2) as the ADC input current is quite low. One easy solution is to add a capacitor across the ADC input and ground as that will [...]

I see. That's clear. What value of capacitor should be used as a capacitor?

G6EJD, 2024-10-26:
T=R.C and aim for 1-sec so 1/47000=0.0000213 or 21uF or actually any value will do, 10uF, 100uF or any value you have hanging around, it's not critical.

AntoszHUN, 2024-10-26:
> T=R.C and aim for 1-sec so 1/47000=0.0000213 or 21uF or actually any value will do, 10uF, 100uF or any value you have hanging around, it's not critical.

I see. Okay. I'll try it.