G6EJD/ESP32-e-Paper-Weather-Display #269: Waveshare 4.2 sketch compilation error

Haranochi, 2024-11-06:
Hi, 

I got this error during compilation, sketch line 208:

'TXT_FORECAST_VALUES' was not declared in this scope. Any help on where I should investigate. 

My conditions:
OS: Linux MX-23
Arduino IDE 1.8.19
ESP32 3.0.7
ArduinoJson 7.2.0
U8g2_for_Adafruit_GFX 1.8.0
GxEPD2 library 1.6.0
Adafruit-GFX-Library 1.11.11
Waveshare 4.2

Best Regards

José Maria

G6EJD, 2024-11-06:
Have you I stalled the supporting library and especially the lang.h file?

Haranochi, 2024-11-06:
Hi,
Thanks for your suggestion, I have checked the lang.h file, this file is not the one in your "src" folder, when I copied yours to the sketch folder it worked. Thanks for your quick reply. Now I have a nice and attractive weather station.
==================================================================================
Feedback from Arduino IDE:
Sketch uses 1152797 bytes (87%) of program storage space. Maximum is 1310720 bytes.
Global variables use 70144 bytes (21%) of dynamic memory, leaving 257536 bytes for local variables. Maximum is 327680 bytes.
Thanks.
Best Regards.
José Maria