G6EJD/ESP32-e-Paper-Weather-Display #270: Board selection

robstermn, 2024-11-09:
Looking for some help with board selection.

I found a LOLIN32 vs LOLIN32D, would the pinout be the same, as I tried with no luck.  And also having issues finding the exact board in VS_code with PlatformIO.

Thanks for any help.

[image]

G6EJD, 2024-11-10:
I've not had much luck with the Heltec boards as the lack of good documentation and the need for a specific Heltec library makes for a complex product. Sone of the best are the Wemos D32 or DFRobot Firebeetle especially the latter, as both can achieve minuscule power consumption in sleep mode, this is where most other development boards fail with often excessive power consumption due to poor design.
I always choose the ESP32 Dev Module as the pins are well marked on the board and as long as you use eg 36 and not VP or D36 then you don't need the compiler to do any pin logical name conversions.

robstermn, 2024-11-10:
> I've not had much luck with the Heltec boards as the lack of good documentation and the need for a specific Heltec library makes for a complex product. Sone of the best are the Wemos D32 or DFRobot [...]

robstermn, 2024-11-10:
Copy that thanks for the recomendation, I did order a Firebeetle. But I did get the LOLIN32 working with Hello World from GXEPD2.

But when I tried to compile your Waveshare 4.2 code , I get the following error, see attached log.

Generating function prototypes...
C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302/bin/xtensa-esp32-elf-g++ -c @C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/flags/defines -IC:\Users\veitchb\AppData\Local\Temp\.arduinoIDE-unsaved20241010-14292-vrikum.nfgai\Waveshare_4_2 -iprefix C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/ @C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/flags/includes -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/qio_qspi/include -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\cores\esp32 -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\variants\esp32 -IC:\Users\veitchb\Documents\Arduino\libraries\ArduinoJson\src -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\WiFi\src -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\Network\src -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SPI\src -IC:\Users\veitchb\Documents\Arduino\libraries\GxEPD2\src -IC:\Users\veitchb\Documents\Arduino\libraries\Adafruit_GFX_Library -IC:\Users\veitchb\Documents\Arduino\libraries\Adafruit_BusIO -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\Wire\src -IC:\Users\veitchb\Documents\Arduino\libraries\U8g2_for_Adafruit_GFX\src -IC:\Users\veitchb\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display-master\src -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\HTTPClient\src -IC:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\NetworkClientSecure\src @C:\Users\veitchb\AppData\Local\Temp\arduino\sketches\8082C3D926E3E40BF90DE84067FC033B/build_opt.h @C:\Users\veitchb\AppData\Local\Temp\arduino\sketches\8082C3D926E3E40BF90DE84067FC033B/file_opts C:\Users\veitchb\AppData\Local\Temp\arduino\sketches\8082C3D926E3E40BF90DE84067FC033B\sketch\Waveshare_4_2.ino.cpp -o C:\Users\veitchb\AppData\Local\Temp\2674805406\sketch_merged.cpp
C:\Users\veitchb\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\veitchb\AppData\Local\Temp\2674805406\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2302/bin/xtensa-esp32-elf-g++" -MMD -c "@C:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32/flags/cpp_flags" -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32/flags/defines" "-IC:\\Users\\veitchb\\AppData\\Local\\Temp\\.arduinoIDE-unsaved20241010-14292-vrikum.nfgai\\Waveshare_4_2" -iprefix "C:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32/include/" "@C:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32/flags/includes" "-IC:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-33fbade6\\esp32/qio_qspi/include" "-IC:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\cores\\esp32" "-IC:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\variants\\esp32" "-IC:\\Users\\veitchb\\Documents\\Arduino\\libraries\\ArduinoJson\\src" "-IC:\\Users\\veitchb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\libraries\\WiFi\\src" "-IC:\\Users\\veitchb\\AppData\\Loca [... cut ...]
Using library HTTPClient at version 3.0.5 in folder: C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\HTTPClient 
Using library NetworkClientSecure at version 3.0.5 in folder: C:\Users\veitchb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\NetworkClientSecure 
exit status 1

Compilation error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?

[error_1110.txt](https://github.com/user-attachments/files/17691980/error_1110.txt)

closed 2025-10-05 by G6EJD