G6EJD/ESP32-e-Paper-Weather-Display #271: Connection failed, error: connection refused

Gorkem-say, 2024-11-22:
Hi, I am trying to upload to my custom esp32-c3 board for e-paper 1.54''. When I upload I got error, what can be the problem?

 serial print::::

Connecting to: INNOVATION
WiFi connected at: 175.110.140.127
12:48:31
connection failed, error: connection refused

G6EJD, 2024-11-22:
This could be an OWM issue as they now charge for API access and charge for excessive use, so your API may be invalid. You'd need to check your account.

Gorkem-say, 2024-11-25:
Thank you for your replies.

I think it is connected to WIFI because when I entered wrong SSID or password, It doesn't connect and doesn't get the time.

Do you suggest any other weather API websites so I can try?

Best regards

G6EJD, 2024-11-25:
I've reviewed the code and the Connection Refused message occurs when it tries to connect to OpenWeatherMapMap, so this is caused by an incorrect API key. You need to log into your OWM account and renew the key then renew in the code to recompile it.
If that doesn't work let me know your key and I'll try it on my browser to debug it for you, you may have to change the code to API 3.0 as 2.5 has been discontinued for some OWM users. Afterwards if you're worried you can always renew your API key.

Or send me via my website Google G6EJD contact me your source code and I'll test it.

G6EJD, 2024-11-25:
http://g6ejd.dynu.com

Gorkem-say, 2024-11-25:
I just sent an email to you..