G6EJD/ESP32-e-Paper-Weather-Display #272: Gray scale vs B/W

robstermn, 2024-11-22:
I got both the 2.9 and the 4.2 working, but the 4.2 seems to be in a gray scale format vs B/W.  Is there anyway to fix that?
see attached.
[image]

Thanks for any help.

G6EJD, 2024-11-22:
Hi Bob, assuming your colour code for black is correct so white is usually all zero bits or as an integer zero and black all the bits are set so; now it can get tricky as an 8-bit data type should be 255, 16-bit 16384 and so-on, so you really need to be using the library colours for black eg EPD_BLACK.

If that doesn't solve the problem then it's mist likely to be a power supply issue or maybe a faulty display.

robstermn, 2024-11-22:
Thanks for the input, I'm trying another esp32.  I took your advise and got a FireBettle2 and I'm trying to figure out the pinouts, and I'm struggling to get this microcontroller to work?  any help with the pinouts would be appricated.

Waveshare epaper to Firebettle2

Busy - 

RST -

DC -

CS -

CLK - 18/sck

DIN - 19/miso

GND - gnd

3.3v - 3.3

https://wiki.dfrobot.com/_SKU_DFR1139_FireBeetle_2_ESP32_E_N16R2_IoT_Microcontroller

G6EJD, 2024-11-22:
I only use the Firebeetle so the standard code pin usage (not Waveshare) applies. DFRobot has a wiki with detailed pin-outs for all their boards, go to their site, choose a product then at the bottom of the page is the link to schematics etc.

closed 2025-10-05 by G6EJD