G6EJD/ESP32-e-Paper-Weather-Display #273: Small error when displaying the night Icons

Zurrmaxe, 2024-11-26:
[image]

Hello Friends
We had clear skies this evening.
The weather station also shows the sun, even though it is dark.
Even with other ICONS, the sun was always visible at night.
To change it, the code must be adjusted.

`void Sunny(int x, int y, bool IconSize, String IconName) {

  int scale = Small;

  if (IconSize == LargeIcon) scale = Large;

  else y = y - 3; 

 **If (IconName.endsWith("n")) {

    addmoon(x, y + 3, scale, IconSize);

  scale = scale * 1.6;

  }

  else  addsun(x, y, scale, IconSize);**

}`

The other ICONS (void XXXXX)  must also be changed in the same way.

With: 
if ( night) and else (day)

Alf

G6EJD, 2024-11-27:
It isn't an error, it's what OpenWeatherMap send out for nighttime conditions!

You need to get OWM to change their API

Zurrmaxe, 2024-11-27:
Ok. But it was funny to see the the sun and moon together at midtnight...I looked at owm and saw, that they use the icon 01n.png for the night and 01d.png for the day.For clean sky only.They use two different Icons, one for day and one night on there website.So i like to use "if then else" to change the Icons when its night.73 Alf

closed 2024-11-27 by G6EJD