G6EJD/ESP32-e-Paper-Weather-Display #283: Several Undefined References ...

teoatess, 2025-09-30:
Building in release mode
Compiling .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o
Linking .pio/build/esp32dev/firmware.elf
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o:(.literal._Z28Convert_Readings_to_Imperialv+0x8): undefined reference to `hPa_to_inHg(float)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o:(.literal._Z28Convert_Readings_to_Imperialv+0xc): undefined reference to `mm_to_inches(float)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o:(.literal._Z8DrawMooniiiii6String+0x18): undefined reference to `NormalizedMoonPhase(int, int, int)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o:(.literal._Z22DisplayForecastSectionii+0x54): undefined reference to `SumOfPrecip(float*, int)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o: in function `Convert_Readings_to_Imperial()':
/Users/toatess/Desktop/Waveshare_Weather/src/forecast_record.h:36: undefined reference to `hPa_to_inHg(float)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/toatess/Desktop/Waveshare_Weather/src/forecast_record.h:36: undefined reference to `mm_to_inches(float)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o: in function `Convert_Readings_to_Imperial()':
/Users/toatess/Desktop/Waveshare_Weather/src/common.h:15: undefined reference to `mm_to_inches(float)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o: in function `DrawMoon(int, int, int, int, int, String)':
/Users/toatess/Desktop/Waveshare_Weather/src/Waveshare_7_5_T7.ino:305: undefined reference to `NormalizedMoonPhase(int, int, int)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/Waveshare_7_5_T7.ino.cpp.o: in function `DisplayForecastSection(int, int)':
/Users/toatess/Desktop/Waveshare_Weather/src/Waveshare_7_5_T7.ino:400: undefined reference to `SumOfPrecip(float*, int)'
/Users/toatess/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/toatess/Desktop/Waveshare_Weather/src/Waveshare_7_5_T7.ino:400: undefined reference to `SumOfPrecip(float*, int)'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1

Am I missing a library?  I am using VSCode with PlatformIO IDE. 

Thanks!

teoatess, 2025-10-01:
I am confused.  There is a solution or no solution?

G6EJD, 2025-10-01:
For your display there is currently no solution. The Wx programme uses the GxEPD2 display driver library which needs to be updated to support the revised display panel drawing mode.

There is nothing I can do.

teoatess, 2025-10-01:
So I purchased all these parts and it won't work?  I printed the casing for it.  I have spent hours and a lot of support to get this to compile. Now it's shit?  Is there another display I can use?

G6EJD, 2025-10-01:
But first have you been through the variants of the 7.5” display objects?

//GxEPD2_BW<GxEPD2_750, GxEPD2_750::HEIGHT> display(GxEPD2_750(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW075T8 640x384, UC8159c (IL0371)

//GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW075T7 800x480, EK79655 (GD7965)

//GxEPD2_BW<GxEPD2_750_GDEY075T7, GxEPD2_750_GDEY075T7::HEIGHT> display(GxEPD2_750_GDEY075T7(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEY075T7 800x480, UC8179 (GD7965)

You may have a T7 chipset, in the main Wx source code there is a display object line, it will be the same as one of these 3 example, try each of the two others, it may solve the problem.

So comment out the current variant with // at the start of the line then paste in one of these at a time, each time compiling to see the end result.

G6EJD, 2025-10-02:
So it's clear, you can only use one of the display object lines, you need to try each in-turn to test if the display driver type will work. I suspect you may have a T7 type of display.
So comment out the line as above that ends with a greyed out image and try one of the others in-turn, only one line in the code.

Waveshare may refer you to their own driver but that doesn't support the GFX drawing library so it's useless.

teoatess, 2025-10-02:
I tried all of the object lines one at a time.  None have worked.

teoatess, 2025-10-02:
> OK, it was worth a try, it illustrates how useless these latest displays are!

should I purchase another display? Perhaps the one I sent you a URL.Thanks,Thomas E Oatess612/850-6671 ***@***.***

G6EJD, 2025-10-03:
It's a gamble either way, if you can find what appears to be old-stock, then it will work.

teoatess, 2025-10-03:
Waveshare sent me a demo to run for the display. It worked flawalessly.  How can I send it to you?

G6EJD, 2025-10-03:
email their files to me? david.g6ejd@gmail.com

It could be their library does not upport the GFX library though.

teoatess, 2025-10-04:
I am not sure what else to do.  Should I look for another weather station to build?  You have any ideas?  I have all these parts they don't work together now.

G6EJD, 2025-10-04:
You should see all the weather data on the serial port.

I am working through the Waveshare library to see if I can simulate the needed FFX commands.

Let's see how I get on.

closed 2025-10-05 by G6EJD